mit-han-lab/tinyengine #2: Tutorial issues

waquey, 2022-09-07:
Thanks for the great work! Currently I do not use ArduCam and I just want to test LCD
After following steps described in the tutorial, I can compile without errors but run with some errors: 

(1) No source available for "d_expression_1() at 0x8001058" 
(2) Even if I move lcdsetup() before "SystemClock_Config()" and try to display string on LCD, it doesn't show anything onto the LCD screen 
(3) Is there any method we can print message in console mode to verify if we enter main() function? 

Could you please help comment that? Thanks

RaymondWang0, 2022-09-07:
Hi @waquey, thanks for your interest in our work! 
For questions 1 and 2, I believe the errors came out due to the lack of ArduCam. To support your request and also other users who do not have an ArduCam, we will revise our tutorial ASAP to support testing TinyEngine without an ArduCam.
At this time, to test TinyEngine without an ArduCam, you could try to comment out the codes related to camera initialization functions in `main.cpp`. Also, please import a testing image as your input image.
For question 3, if you are using STM32CubeIDE, you can just get into the Debug Mode in the IDE, and use its built-in functions (such as adding breakpoints or using `Run to Line` function). After setting up and starting the debug process in the Debug Mode, you should be able to verify whether you enter `main()` function or not.
For question 4, yes, please use `HAL_GetTick()` before and after `invoke()` to obtain the inference latency.

If you have any further questions, please feel free to let us know!

RaymondWang0, 2022-09-07:
Hi @waquey, thanks for your interest in our work!
For questions 1 and 2, I believe the errors came out due to the lack of ArduCam. To support your request and also other users who do not have an ArduCam, we will revise our tutorial ASAP to support testing TinyEngine without an ArduCam.
At this time, to test TinyEngine without an ArduCam, you could try to comment out the codes related to camera initialization functions in main.cpp. Also, please import a testing image as your input image.
For question 3, if you are using STM32CubeIDE, you can just get into the Debug Mode in the IDE, and use its built-in functions (such as adding breakpoints or using Run to Line function). After setting up and starting the debug process in the Debug Mode, you should be able to verify whether you enter main() function or not.
For question 4, yes, please use HAL_GetTick() before and after invoke() to obtain the inference latency.

If you have any further questions, please feel free to let us know!

meenchen, 2022-12-08:
Close due to inactivity. Feel free to reopen.

closed 2022-12-08 by meenchen