mit-han-lab/tinyengine #3: OpenMV Entegration Tutorial

BehicKlncky, 2022-09-07:
Anyone knows that how to entegrate this engine to OpenMV H7 Plus board ?

meenchen, 2022-09-07:
Hi @BehicKlncky, thanks for your interest in our work! To integrate Tinyengine to OpenMV H7 Plus, you will need to modify the firmware of OpenMV and create a Python API for Tinyengine. Please refer to their GitHub repo: https://github.com/openmv/openmv/tree/master/src to build the firmware from the source. If you have further questions, feel free to let me know.

BehicKlncky, 2022-10-15:
Hi @meenchen ,
Actually, integration of your engine to my OpenMV H7 Plus Board was the last step of my whole project, which I decided to start when you released this engine. Now, I've been studying on my project and I'm about the finish which means I came to the last step of it: integration of the engine to the board. 
So, because of the fact that I should not be considered as a professional on microcontrollers, a little further help would help me a lot. What did you exactly mean by "modifying the firmware" and like how exactly a Python API will I need? If you have an enough time to explain a little detailed for me, I'd be really appreciate it. Thanks a lot for your help.

senceryucel, 2022-10-15:
Trying to deal with the same thing. If it is possible, it'd be great to hear from you guys about the integration process of the engine to the OpenMV H7 Plus.

Appreciate your work, thanks a lot in advance!

meenchen, 2022-12-08:
Hi @BehicKlncky and @senceryucel,

Recently, we release a tutorial for deploying the VWW demo on the OpenMV cam. Feel free to try it out.

Link of the tutorial: https://github.com/mit-han-lab/tinyengine/tree/master/examples/openmv_vww

meenchen, 2023-02-06:
Close due to inactivity. Feel free to reopen.

closed 2023-02-06 by meenchen

senceryucel, 2023-03-11:
Hello again, @meenchen 

@BehicKlncky and I were trying to integrate this beautiful engine into our H7 Plus Board. Everything seemed to be working fine, until we tried to run the example person detection script. 

`
import cexample
import sensor

sensor.reset()  # Reset and initialize the sensor.
sensor.set_pixformat(sensor.RGB565)  # Set pixel format to RGB565 (or GRAYSCALE)
sensor.set_framesize(sensor.HD)  # Set frame size to QVGA 160x128

while True:
    img = sensor.snapshot()  # Take a picture and return the image.
    ret = cexample.person_detection(img, 0.15)
`

Since we have no LCD screen, we just modified the code a little to test it via frame buffer. However, when we run, it says that there is no module named "cexample".

We looked a bit in the repo for it, but could not find anything that would help us. Any help will be appreciated. Thanks a lot in advance!

senceryucel, 2023-03-11:
I decided to open a new issue about it since the problem I am facing with is not directly connected with the problem in the header of this issue.