mit-han-lab/tinyengine #8: visual wake word (VWW) model end to end workflow?

ramkumarkoppu, 2022-09-20:
Hi,

Where do I find jupyter notebook of given example visual wake word (VWW) model in the tutorial folder if I want to check it's model architecture and how it is optimized for MCU (pruned, quantized, model conversion (how C/C++ code is generated from the model)) to follow end to end workflow?

meenchen, 2022-09-24:
Hi @ramkumarkoppu, thanks for your interest in our work. For the model optimization, please refer to our [mcunet repo.](https://github.com/mit-han-lab/mcunet) We also have a model zoo where you can download the pre-trained models and check their model architecture. Then, you can follow these [examples](https://github.com/mit-han-lab/tinyengine/tree/master/examples) to generate C/C++ code.

ramkumarkoppu, 2022-09-24:
Hi @meenchen, Do we have corresponding jupyter notebook for vww model? if yes, where do I find it.

tonylins, 2022-09-24:
Hi, currently we do not have a jupyter notebook for the example. But I think the provided code works the same.

ramkumarkoppu, 2022-09-28:
Hi, I am interested to understand how fp32 PyTorch model is pruned, quantised before converting into C code. Where do I find the relevant code in the repo?

meenchen, 2022-12-20:
Close due to inactivity. Feel free to reopen.

closed 2022-12-20 by meenchen