mit-han-lab/tinyengine #10: vvw.py error

KwangryeolPark, 2022-10-01:
Hello, I'm trying to do exmaple. However I encounter this error while executing "python vvw.py"

```
(pytorch) kwangryeol@kwangryeol-OMEN-30L-Desktop-GT13-1xxx:~/바탕화면/tinyengine$ python examples/vww.py 
Deriving the memory schedule for 41 activation tensors.
100%|██████████████████████████████████████████████████████████████████████████| 41/41 [00:00<00:00, 185109.22it/s]
Traceback (most recent call last):
  File "/home/kwangryeol/바탕화면/tinyengine/examples/vww.py", line 28, in <module>
    peakmem = GenerateSourceFilesFromTFlite(
  File "/home/kwangryeol/바탕화면/tinyengine/code_generator/CodegenUtilTFlite.py", line 54, in GenerateSourceFilesFromTFlite
    memory_scheduler.allocateMemory()
  File "/home/kwangryeol/바탕화면/tinyengine/code_generator/GeneralMemoryScheduler.py", line 190, in allocateMemory
    self.allocator.visualize(self.mem_visual_path)
  File "/home/kwangryeol/바탕화면/tinyengine/code_generator/allocator/base_allocator.py", line 240, in visualize
    plt.savefig(path, dpi=FIGURE_CONFIG["DPI"])
  File "/home/kwangryeol/anaconda3/envs/pytorch/lib/python3.9/site-packages/matplotlib/pyplot.py", line 942, in savefig
    res = fig.savefig(*args, **kwargs)
  File "/home/kwangryeol/anaconda3/envs/pytorch/lib/python3.9/site-packages/matplotlib/figure.py", line 3272, in savefig
    self.canvas.print_figure(fname, **kwargs)
  File "/home/kwangryeol/anaconda3/envs/pytorch/lib/python3.9/site-packages/matplotlib/backend_bases.py", line 2338, in print_figure
    result = print_method(
  File "/home/kwangryeol/anaconda3/envs/pytorch/lib/python3.9/site-packages/matplotlib/backend_bases.py", line 2204, in <lambda>
    print_method = functools.wraps(meth)(lambda *args, **kwargs: meth(
  File "/home/kwangryeol/anaconda3/envs/pytorch/lib/python3.9/site-packages/matplotlib/_api/deprecation.py", line 410, in wrapper
    return func(*inner_args, **inner_kwargs)
  File "/home/kwangryeol/anaconda3/envs/pytorch/lib/python3.9/site-packages/matplotlib/backends/backend_agg.py", line 520, in print_png
    self._print_pil(filename_or_obj, "png", pil_kwargs, metadata)
  File "/home/kwangryeol/anaconda3/envs/pytorch/lib/python3.9/site-packages/matplotlib/backends/backend_agg.py", line 466, in _print_pil
    FigureCanvasAgg.draw(self)
  File "/home/kwangryeol/anaconda3/envs/pytorch/lib/python3.9/site-packages/matplotlib/backends/backend_agg.py", line 408, in draw
    self.figure.draw(self.renderer)
  File "/home/kwangryeol/anaconda3/envs/pytorch/lib/python3.9/site-packages/matplotlib/artist.py", line 74, in draw_wrapper
    result = draw(artist, renderer, *args, **kwargs)
  File "/home/kwangryeol/anaconda3/envs/pytorch/lib/python3.9/site-packages/matplotlib/artist.py", line 51, in draw_wrapper
    return draw(artist, renderer)
  File "/home/kwangryeol/anaconda3/envs/pytorch/lib/python3.9/site-packages/matplotlib/figure.py", line 3069, in draw
    mimage._draw_list_compositing_images(
  File "/home/kwangryeol/anaconda3/envs/pytorch/lib/python3.9/site-packages/matplotlib/image.py", line 131, in _draw_list_compositing_images
    a.draw(renderer)
  File "/home/kwangryeol/anaconda3/envs/pytorch/lib/python3.9/site-packages/matplotlib/artist.py", line 51, in draw_wrapper
    return draw(artist, renderer)
  File "/home/kwangryeol/anaconda3/envs/pytorch/lib/python3.9/site-packages/matplotlib/axes/_base.py", line 3099, in draw
    self.patch.draw(renderer)
  File "/home/kwangryeol/anaconda3/envs/pytorch/lib/python3.9/site-packages/matplotlib/artist.py", line 51, in draw_wrapper
    return draw(artist, renderer)
  File "/home/kwangryeol/anaconda3/envs/pytorch/lib/python3.9/site-packages/matplotlib/patches.py", line 589, in draw
    self._draw_paths_with_artist_properties(
  File "/home/kwangryeol/anaconda3/envs/pytorch/lib/python3.9/site-packages/matplotlib/patches.py", line 574, in _draw_paths_with_artist_properties
    renderer.draw_path(gc, *draw_path_args)
  File "/home/kwangryeol/anaconda3/envs/pytorch/lib/python3.9/site-packages/matplotlib/backends/backend_agg.py", line 149, in draw_path
    self._renderer.draw_path(gc, path, transform, rgbFace)
TypeError: must be real number, not str
```

KwangryeolPark, 2022-10-01:
I fix the error.

closed 2022-10-01 by KwangryeolPark

weitung, 2022-11-08:
I'm encountering the same bug, can you tell me how you fixed it?

KwangryeolPark, 2022-11-09:
> I'm encountering the same bug, can you tell me how you fixed it?

See Setup for Users and Setup for Developers section in README.md file. 😁