mit-han-lab/tinyengine #14: blank screen

jonathan-ko, 2022-11-06:
Hi, I built the TinyEngine tutorial with zero errors. Then, after the download, the screen goes black immediately. No errors are mentioned. No warnings either. Any thoughts or advice?

[image]

liuruiqi0520, 2022-11-07:
I had a similar problem

RaymondWang0, 2022-11-07:
Hi @jonathan-ko and @liuruiqi0520, here are the debugging suggestions for your problems:
1) Please check whether you download the correct version (1.5.0) of STM32CubeIDE for the tutorial.
2) Please check whether you have the [same equipment](https://github.com/mit-han-lab/tinyengine/tree/master/tutorial#equipments) as ours.
3) Please check whether your PIN connection between Arducam and the board is correct ([Step 4 in the tutorial](https://github.com/mit-han-lab/tinyengine/tree/master/tutorial#detailed-instruction)).

jonathan-ko, 2022-11-12:
Thank you "download the correct version (1.5.0)" was my problem -- I downloaded the newest 1.10 -- warning to others.

tonylins, 2022-11-14:
I will close the issue since the problem is solved. Feel free to reopen.

closed 2022-11-14 by tonylins

liuruiqi0520, 2022-12-02:
Hello! Your answer perfectly solves my problem. I impressed by your excellent work in the field of TinyML and studied the source code of MCUNet and Once-for-all, But when I try to replicate TinyNAS with the method in the Once-for-all, I can't find the pre training weight of the supernet used in MCUNet, could you please release the pre training weight? It would be very helpful for my next work and I would greatly appreciate if you could reply me.

抄送: "( /。\ ***@***.******@***.***&gt;;