mit-han-lab/tinyengine #18: mcunet-10fps inference failing shows nonperson always

brainchip-india, 2022-11-19:
With the help of the [tinyengine/tutorial](https://github.com/mit-han-lab/tinyengine/tree/master/tutorial) I am able to run the mcunet_5ps(default) and able to see the inference is working fine, so tried changing model to mcunet_10fps in tinyengine/tutorial/examples/vww.py file able to run the code but the inference is failing (always showing no person though the person is present).

anything I need to take care to run mcunet_10fps? 
@meenchen @RaymondWang0 please help, thanks in advance.

[image]

brainchip-india, 2022-11-22:
@meenchen @RaymondWang0 not sure you both have seen the issue, kindly reminding the issue.

meenchen, 2022-11-22:
Hi @brainchip-india,

In the tutorial, the application code assumes the resolution of the model is 80x80, but for mcunet_10fps, it is actually 64x64. You will need to change the data loading part in the main.cpp for different models.

brainchip-india, 2022-11-23:
Thanks @meenchen it worked, it's better you log it somewhere in the Readme for newbies like me.

closed 2022-12-05 by brainchip-india