mit-han-lab/tinyengine #20: VWW demo and Arducam Shield Mini 2MP Plus

ahdxwg, 2022-11-28:
I am very interested in your project and want to realize it.    I couldn't get the Arducam Shield Mini **2MP** Plus on Taobao.Can I use Arducam Shield Mini **5MP** Plus?

ahdxwg, 2022-11-29:
[image]
Could I use DEV-18440 ARDUCAM 5MP PLUS OV5642 MINI CAM ?Thanks

meenchen, 2022-12-08:
Hi @ahdxwg,

Unfortunately, our demo only works with the 2MP camera, but recently we have released another [tutorial for OpenMV cam](https://github.com/mit-han-lab/tinyengine/tree/master/examples/openmv_vww). You should be able to get one OpenMV cam easily.

meenchen, 2023-02-06:
Close due to inactivity. Feel free to reopen.

closed 2023-02-06 by meenchen