mit-han-lab/tinyengine #27: Code Generation Patch Based Inference Bug

Zerkclown, 2022-12-12:
Facing an issue related to PR #26:

> Traceback (most recent call last):
  File "examples/vww.py", line 31, in <module>
    life_cycle_path="./lifecycle.png",
  File "/Users/amahmed/Desktop/UMass/Spring_2022/Thesis/tinyengine/code_generator/CodegenUtilTFlite.py", line 70, in GenerateSourceFilesFromTFlite
    code_generator.codeGeneration()
  File "/Users/amahmed/Desktop/UMass/Spring_2022/Thesis/tinyengine/code_generator/CodeGenerator.py", line 131, in codeGeneration
    self._genPatchInference()
  File "/Users/amahmed/Desktop/UMass/Spring_2022/Thesis/tinyengine/code_generator/CodeGenerator.py", line 182, in _genPatchInference
    last_patch_op_output_buffer_str_for_patch_inference = last_patch_op._getBufferstr(
AttributeError: 'NoneType' object has no attribute '_getBufferstr'

Deleting lines 179 to 181 in `CodeGenerator.py` removes the error.

meenchen, 2022-12-12:
Hi @Zerkclown,

Thanks for reporting the bug. The fix has been merged: https://github.com/mit-han-lab/tinyengine/pull/28.

Zerkclown, 2022-12-12:
Thank you, this fixed it!