mit-han-lab/tinyengine #31: IDE Compilation Error

Zerkclown, 2022-12-13:
Currently following the [inference tutorial](https://github.com/mit-han-lab/tinyengine/tree/master/tutorial/inference) and attempting to build the the VWW demo (step 3 of the tutorial). STM32CubeIDE seems to be unable to fully compile the project. Any suggestions to where the source of the error might lie?

Is it in the Makefile? or could the error have been caused all the way back during the CodeGeneration step? Am I including the wrong version of MCUNet?

[image]

Zerkclown, 2022-12-13:
The issue seems to be caused by incongruence of `patchpadding_convolve_s8_kernel3_inputch3_stride2` declaration's and implementation's parameters. An additional parameter (`q15_t *kbuf`) can be found in the declaration:

https://github.com/mit-han-lab/tinyengine/blob/3e6292e7ab5c24b3011856fac2830863164f2fea/TinyEngine/include/tinyengine_function.h#L154

meenchen, 2022-12-14:
Hi @Zerkclown,

Thanks for fixing the issue. I will close this issue since the fix has been merged.

closed 2022-12-14 by meenchen