mit-han-lab/tinyengine #33: Demo without LCD/Camera

irvingzhang0512, 2022-12-15:
Thanks for the great work!

For now, the demos are all strongly related to the camera and LCD. I think a demo without camera/lcd, and read image from head file and print output results will be very helpful.

BTW: I'm working on this demo in spare time, but I'm familiar with stm32... Don't know how much time I need to finish this

meenchen, 2022-12-19:
Hi @irvingzhang0512,

Thanks for reaching out. You can use the `TESTTENSOR` macro to remove the dependency on LCD/Camera. https://github.com/mit-han-lab/tinyengine/blob/03639563ebf6538fff557515e31667fca6448cd3/tutorial/inference/Src/main.cpp#L80

To set the input image,  please refer to the following code block
https://github.com/mit-han-lab/tinyengine/blob/03639563ebf6538fff557515e31667fca6448cd3/tutorial/inference/Src/main.cpp#L111

Hope this could help you achieve your goal. Let me know if you have further questions.

meenchen, 2023-02-06:
Close due to inactivity. Feel free to reopen.

closed 2023-02-06 by meenchen