mit-han-lab/tinyengine #35: Would you mind you uploading inplace depthwise kernel file on github?

asdfrv100, 2023-01-02:
Hi. Thank you for your contribution.

I generated c++ code of Mobilent V2(net id=mbv2-320kB) model with patch-based option by tiny engine.

when I did end-to-end working test in stm32ide, it's not working with error message that undefined reference to `depthwise_kernel7x7_stride2_inplace_CHW_fpreq'

I spend a lot of time finding this file but I could't.

Would you mind you uploading these file on github?

************error message***********

11:21:03 **** Incremental Build of configuration Debug for project mbv2_patch_1226 ****
make -j8 all 
arm-none-eabi-g++ -o "mbv2_patch_1226.elf" @"objects.list"   -mcpu=cortex-m7 -T"../STM32F746NGHx_FLASH.ld" --specs=nosys.specs -Wl,-Map="mbv2_patch_1226.map" -Wl,--gc-sections -static  -mfpu=fpv5-sp-d16 -mfloat-abi=hard -mthumb -Wl,--start-group -lc -lm -lstdc++ -lsupc++ -Wl,--end-group
c:\st\stm32cubeide_1.5.0\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.7-2018-q2-update.win32_1.5.0.202011040924\tools\arm-none-eabi\bin\ld.exe:../STM32F746NGHx_FLASH.ld:163: warning: memory region `DTCMRAM' not declared
Src/TinyEngine/codegen/Source/genModel.o: In function `invoke':
/Users/raymondwang/STM32CubeIDE/workspace_1.5.0/TinyEngine_vww_tutorial/Debug/../Src/TinyEngine/codegen/Source/genModel.c:62: undefined reference to `depthwise_kernel7x7_stride2_inplace_CHW_fpreq'
/Users/raymondwang/STM32CubeIDE/workspace_1.5.0/TinyEngine_vww_tutorial/Debug/../Src/TinyEngine/codegen/Source/genModel.c:76: undefined reference to `depthwise_kernel5x5_stride1_inplace_CHW_fpreq'
/Users/raymondwang/STM32CubeIDE/workspace_1.5.0/TinyEngine_vww_tutorial/Debug/../Src/TinyEngine/codegen/Source/genModel.c:84: undefined reference to `depthwise_kernel7x7_stride2_inplace_CHW_fpreq'
/Users/raymondwang/STM32CubeIDE/workspace_1.5.0/TinyEngine_vww_tutorial/Debug/../Src/TinyEngine/codegen/Source/genModel.c:90: undefined reference to `depthwise_kernel7x7_stride1_inplace_CHW_fpreq'
/Users/raymondwang/STM32CubeIDE/workspace_1.5.0/TinyEngine_vww_tutorial/Debug/../Src/TinyEngine/codegen/Source/genModel.c:112: undefined reference to `depthwise_kernel5x5_stride2_inplace_CHW_fpreq'
/Users/raymondwang/STM32CubeIDE/workspace_1.5.0/TinyEngine_vww_tutorial/Debug/../Src/TinyEngine/codegen/Source/genModel.c:134: undefined reference to `depthwise_kernel7x7_stride1_inplace_CHW_fpreq'
collect2.exe: error: ld returned 1 exit status
make: *** [makefile:88: mbv2_patch_1226.elf] Error 1
"make -j8 all" terminated with exit code 2. Build might be incomplete.

11:21:05 Build Failed. 7 errors, 1 warnings. (took 2s.39ms)

**********************end error message ************************

Once again, Thank you for your work.

meenchen, 2023-01-03:
Hi @asdfrv100,

Thanks for your interest in our work. That file should be generated automatically during code generation. Could you check if that file is under the `codegen/Source`.

asdfrv100, 2023-01-10:
HI meenchen. Thanks for your answer.

I'm sorry for my answer too late.

I checked codegen/Source file problem that you talked. but there is no problem.

I solved my problem. that was my fault about linking debug/***.elf file.

Project folder name must be matched with .elf file name. if not, debug file's link is not updated.(for example, /Users/raymondwang/STM32CubeIDE/workspace_1.5.0)

meenchen, 2023-01-10:
Glad to hear your problem is solved. I will close this issue, and please feel free to reopen it if needed.

closed 2023-01-10 by meenchen