mit-han-lab/tinyengine #40: mcunet face detection model and code

kumarchaitanyab, 2023-01-25:
Hi Team,  I would like to run the mcunet timyengine face detection model on M7 board and looking to find the source code but I see that there is only VWW code captured here. Is there a way to find the face detection code also?

Thanks in advance.

meenchen, 2023-01-27:
Hi @kumarchaitanyab, we have an example for person detection: https://github.com/mit-han-lab/tinyengine/blob/master/examples/detection_fpn.py. Feel free to check it out.

kumarchaitanyab, 2023-01-29:
Thanks for your response. Actually I am looking for  C/C++ source code for face detection based application which runs on M7 based microcontroller. Can you please point me to that?

meenchen, 2023-02-09:
Hi @kumarchaitanyab,

Sorry for the late reply. We are preparing to release the source code and tutorial for the face mask detection demo, so please stay tuned.

meenchen, 2023-02-10:
Hi @kumarchaitanyab,

We have released the source code for face mask detection. Feel free to check it out and let us know any issues or questions.

Tutorial on openmv cam: https://github.com/mit-han-lab/tinyengine/tree/master/examples/openmv_face_mask_detection
Generated code for the model: https://github.com/mit-han-lab/tinyengine/tree/master/examples/openmv_face_mask_detection/codegen

kumarchaitanyab, 2023-02-15:
Sure, will try and let you know. However, can you share your benchmarking numbers for various boards, more specifically am looking for detection timings. I have an M7 (STM32F746G-DISCO) board and hope this code can run on this board.

kumarchaitanyab, 2023-02-15:
Hi @meenchen @RaymondWang0 

I have arducam, not sure if this code supports this camera interface. Can you please confirm and let me know along with benchmarking numbers for face detection and specially for STM32F746G-DISCO board.

meenchen, 2023-02-21:
Hi @kumarchaitanyab,

The generated code for Tinyengine is able to run on the STM32F746G-DISCO board, but the application-specific code will need to be edited (refer to tutorial). For face mask detection, the latency is about 140ms on the openmv cam running at 480MHz. On the STM32F746G-DISCO board, the latency will be around 311ms at 216MHz.

kumarchaitanyab, 2023-02-23:
Hi @meenchen 
Thanks for the input. Just wanted to know that this time (140ms or 311ms) is only for inference or does it includes camera capture and pre, post processing involved?

meenchen, 2023-02-23:
> Hi @meenchen Thanks for the input. Just wanted to know that this time (140ms or 311ms) is only for inference or does it includes camera capture and pre, post processing involved?

That includes post processing as well.

kumarchaitanyab, 2023-02-24:
Hi @meenchen, thanks a lot. 
It will be very useful for us if you can provide timing breakdown for camera capture, pre-processing, inference and post-processing.