mit-han-lab/tinyengine #41: Would you mind upload some kernel of tiny engine?

asdfrv100, 2023-01-26:
Hello again, and Thanks for your work.

I found that there is no kernel, patchpadding_depthwise_kernel5x5_stride1_inplace_CHW.c, 
patchpadding_depthwise_kernel5x5_stride2_inplace_CHW.c,
and 7x7 stride1 and 2..so on.

file location tinyengine/TinyEngine/src/kernels/int_forward_op.

Would you mind upload these kernel of tiny engine?

If that's impossible, I'd appreciate it if you could let me know as well.

.......................................................................................................................................................

Recently, I successed your Patch-based code genertion(with mcunet github's model-zoo tflite file), and it worked well with some model, on my board. (I solved a lot of error for this work, but anyway it's working.)

I did the end-to-end test on stm32 F746gz-disc board, and camera.(same stuff as the tutorial) 

but I found that there is no kernel, patch_depthwise_kernel5x5_stride1_inplace_kernel_CHW.c and patch_depthwise_kernel7x7_stride1_inplace_kernel_CHW.c  file in project forder path, 
tinyengine/TinyEngine/src/kernels/int_forward_op.

***my Experiment result

(this Experisment is wrong because I changed patchpadding file from 5x5 to 3x3.) 

[image]
[image]
[image]

once again, Thanks for your work.

meenchen, 2023-01-27:
Hi @asdfrv100, unfortunately, we don't have the implementation for 5x5 or 7x7 kernels yet. For patch-based inference, we prefer using a smaller kernel to reduce the receptive field.

asdfrv100, 2023-02-03:
hi@meenchen , Thank's for answer. :) 

I have a one more question. I want to know how to make MBV2 model for F746gz-disc board and Tinyengine.

I generated Mbv2 model by tensorflow library, but it is too big to use in my board.

I know that paper'model be used QAT, but I need other information for model generation.

Here is my model's inforamition. After learning fp32 MBV2 for only 1 epoch on the fp32 Imagenet-1000 dataset,I used post training quantization to quantize the model with int8 weights and activations. (but 2.2MB)

Could you please tell me what I need to do to make this model less than 1 MB like the MCUNET paper?

*board information
[image]

*my model information
[image]

once again, Thank's your work.

meenchen, 2023-02-08:
Hi @asdfrv100,

You will need to shrink down the width of the model (e.g., our benchmark uses 0.35) and change the number of output classes to a smaller setting.

asdfrv100, 2023-02-11:
Hi @meenchen,

I checked the structure of the benchmark model you mentioned using https://netron.app/, but the output that passed the last conv operation had 1x1000 classes. In addition, we understand that in the paper experiment Fig 6 of MCUNetV2, a model with a width of 0.5 was used. Isn't that another reason?

Below is the result of checking the structure of the mbV2w0.35r144 model on the site.

[image]

and I uploaded my model tfml file and tensorflow code.

*tfml model file
[mobilenet_v2_quantized.zip](https://github.com/mit-han-lab/mcunet/files/10706935/mobilenet_v2_quantized.zip)

* Here is my model generation code

<details>
<summary>more</summary>

import os
import tensorflow as tf
import config as c
from tqdm import tqdm
from tensorflow.keras import optimizers
from utils.data_utils import train_iterator
from utils.eval_utils import cross_entropy_batch, correct_num_batch, l2_loss
from tensorflow.keras.applications.mobilenet_v2 import MobileNetV2
from model.ResNet import ResNet
from model.ResNet_v2 import ResNet_v2
from test import test
import numpy as np
import os
import PIL
import PIL.Image
import tensorflow as tf
import pathlib

def representative_data_gen():
        for input_value in tf.data.Dataset.from_tensor_slices(train_ds).batch(1).take(100):
                #Model has only one input so each data point has one element.
                 yield [input_value]

def representative_dataset():
     for _ in range(100):
        data = np.random.rand(1, 144, 144, 3)
        yield [data.astype(np.float32)]

os.environ['CUDA_VISIBLE_DEVICES'] = '0'
os.environ['CUDA_VISIBLE_DEVICES'] = '1'

checkpoint_path = "training_1/cp.ckpt"
checkpoint_dir = os.path.dirname(checkpoint_path)

data_dir = pathlib.Path("/home/poweroverwhelming/modelgen/ImageNet_ResNet_Tensorflow2.0/imagenet10/train")

image_count = len(list(data_dir.glob('*/*.JPEG')))

#training param
batch_size = 32
img_height = 144
img_width = 144
num_classes =1000

#directroy making
train_ds = tf.keras.utils.image_dataset_from_directory(
        data_dir,
        validation_split=0.2,
        subset="training",
        seed=123,
        image_size=(img_height, img_width),
        batch_size=batch_size)

val_ds  = tf.keras.utils.image_dataset_from_directory(
         data_dir,
         validation_split=0.2,
         subset="validation",
         seed=123,
         image_size=(img_height, img_width),
         batch_size=batch_size)

#print class name

#nomalization
normalization_layer = tf.keras.layers.Rescaling(1./255)

normalized_ds = train_ds.map(lambda x, y: (normalization_layer(x), y))
image_batch, labels_batch = next(iter(normalized_ds))
first_image = image_batch[0]
#Notice the pixel values are now in `[0,1]`.
#print(np.min(first_image), np.max(first_image))

AUTOTUNE = tf.data.AUTOTUNE

train_ds = train_ds.cache().prefetch(buffer_size=AUTOTUNE)
val_ds = val_ds.cache().prefetch(buffer_size=AUTOTUNE)

#training
model = MobileNetV2(input_shape=(144, 144, 3), alpha=0.5, weights=None, classes=1000)
model.compile(
          optimizer='adam',
            loss=tf.keras.losses.SparseCategoricalCrossentropy(from_logits=True),
              metrics=['accuracy'])

cp_callback = tf.keras.callbacks.ModelCheckpoint(filepath=checkpoint_path,
                                                 save_weights_only=True,
                                                 verbose=1)

model.fit(train_ds,
          validation_data=val_ds,
          epochs=50)

#loss, acc = model.evaluate(test_images, test_labels, verbose=2)
#print("Untrained model, accuracy: {:5.2f}%".format(100 * acc))

#Save the entire model as a SavedModel.
#!mkdir -p saved_model
model.save('saved_model/my_model')

#PTQ part
#Convert the model to TensorFlow Lite format
converter = tf.lite.TFLiteConverter.from_saved_model('saved_model/my_model')

#Perform post-training quantization
converter.optimizations = [tf.lite.Optimize.DEFAULT]
converter.target_spec.supported_ops = [tf.lite.OpsSet.TFLITE_BUILTINS_INT8]
converter.inference_input_type = tf.int8
converter.inference_output_type = tf.int8
converter.representative_dataset = representative_dataset

#Save the quantized model
tflite_model = converter.convert()
open("mobilenet_v2_imgnet10_r144_w05_quantizaed.tflite", "wb").write(tflite_model)

</details>

Thank you.

meenchen, 2023-02-15:
> 

Hi @asdfrv100,

I compare your model with the vanilla mbv2, but the width multiplier does not seem 0.5 to me. E.g.,:
[image]

meenchen, 2023-04-18:
Close due to inactivity. Feel free to reopen.

closed 2023-04-18 by meenchen