mit-han-lab/tinyengine #46: inference tutorial error

Super-Karl, 2023-02-08:
> 14:35:33 **** Incremental Build of configuration Debug for project TinyEngine_vww_tutorial ****
make -j7 all 
arm-none-eabi-g++ -o "TinyEngine_vww_tutorial.elf" @"objects.list"   -mcpu=cortex-m7 -T"../STM32F746NGHx_FLASH.ld" --specs=nosys.specs -Wl,-Map="TinyEngine_vww_tutorial.map" -Wl,--gc-sections -static  -mfpu=fpv5-sp-d16 -mfloat-abi=hard -mthumb -Wl,--start-group -lc -lm -lstdc++ -lsupc++ -Wl,--end-group
/Applications/STM32CubeIDE.app/Contents/Eclipse/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.7-2018-q2-update.macos64_1.5.0.202011040924/tools/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld:../STM32F746NGHx_FLASH.ld:163: warning: memory region `DTCMRAM' not declared
Src/TinyEngine/src/kernels/fp_requantize_op/convolve_1x1_s8_ch16_fpreq.o: In function `convolve_1x1_s8_ch16_fpreq':
/Users/karl/STM32CubeIDE/workspace_1.5.0/TinyEngine_vww_tutorial/Debug/../Src/TinyEngine/src/kernels/fp_requantize_op/convolve_1x1_s8_ch16_fpreq.c:61: undefined reference to `write_q15x2_ia'
/Users/karl/STM32CubeIDE/workspace_1.5.0/TinyEngine_vww_tutorial/Debug/../Src/TinyEngine/src/kernels/fp_requantize_op/convolve_1x1_s8_ch16_fpreq.c:61: undefined reference to `write_q15x2_ia'
/Users/karl/STM32CubeIDE/workspace_1.5.0/TinyEngine_vww_tutorial/Debug/../Src/TinyEngine/src/kernels/fp_requantize_op/convolve_1x1_s8_ch16_fpreq.c:62: undefined reference to `write_q15x2_ia'
/Users/karl/STM32CubeIDE/workspace_1.5.0/TinyEngine_vww_tutorial/Debug/../Src/TinyEngine/src/kernels/fp_requantize_op/convolve_1x1_s8_ch16_fpreq.c:62: undefined reference to `write_q15x2_ia'
/Users/karl/STM32CubeIDE/workspace_1.5.0/TinyEngine_vww_tutorial/Debug/../Src/TinyEngine/src/kernels/fp_requantize_op/convolve_1x1_s8_ch16_fpreq.c:88: undefined reference to `write_q15x2_ia'
Src/TinyEngine/src/kernels/fp_requantize_op/convolve_1x1_s8_ch16_fpreq.o:/Users/karl/STM32CubeIDE/workspace_1.5.0/TinyEngine_vww_tutorial/Debug/../Src/TinyEngine/src/kernels/fp_requantize_op/convolve_1x1_s8_ch16_fpreq.c:88: more undefined references to `write_q15x2_ia' follow
collect2: error: ld returned 1 exit status
make: *** [makefile:88: TinyEngine_vww_tutorial.elf] Error 1
"make -j7 all" terminated with exit code 2. Build might be incomplete.
14:35:34 Build Failed. 7 errors, 1 warnings. (took 468ms)

I followed your tutorial, using macOS, python 3.6, stm32cubeide 1.5.0, but got this error. Please help to fix it. Thank you very much!!!

Super-Karl, 2023-02-08:
debug:
add `#include "arm_math_memory.h"` in file `tutorial/TinyEngine_vww_tutorial/Src/TinyEngine/include/img2col_element.h` at line 23.

meenchen, 2023-02-08:
Hi @Super-Karl,

Thanks for reporting this issue. I have pushed the fix: https://github.com/mit-han-lab/tinyengine/pull/47.

Super-Karl, 2023-02-12:
Thank you, very much. @meenchen

closed 2023-02-14 by meenchen