mit-han-lab/tinyengine #49: Cannot run Codegen to generate code for other models

729557989, 2023-02-09:
I was trying to deploy a model with a different input shape to the STM32 board, but running this command raises **NotImplementedError**:

python examples/tiny_training.py -f full_bp-1x3x128x128-graph.json -D full_bp-1x3x128x128-params.pkl -QAS scale.json -m -g -d -FR

Where **scale.json** comes from img1 (highlighted)
And both **full_bp-1x3x128x128-graph.json** & **full_bp-1x3x128x128-params.pkl** comes from img2 (highlighted)

These 3 files were generated accordingly from the tiny_training repo's compilation/readme.md

Any thoughts for this issue?

[image]
[image]

meenchen, 2023-02-09:
Hi @729557989,

Thanks for reporting this issue. Could you also share the error message and stack trace for the issue?

729557989, 2023-02-10:
Does this help?

**Commands I Ran:**
1. conda activate tinyengine
2. export PYTHONPATH=${PYTHONPATH}:$(pwd)
3. python examples/tiny_training.py -f sm_assets/full_bp-1x3x128x128-graph.json -D sm_assets/full_bp-1x3x128x128-params.pkl -QAS sm_assets/scale.json -m -g -d -FR

_Note: the 3 files accessed from 3. were all generated  accordingly from the mit-han-lab/tiny_training repo's compilation/readme.md_

**The Error Message + Stack Trace:**

Traceback (most recent call last):
  File "examples/tiny_training.py", line 115, in < module >
    main()
  File "examples/tiny_training.py", line 72, in main
    parser.loadModel()
  File "/home/jimmy/low_power/tinyengine/code_generator/TTEParser.py", line 438, in loadModel
    raise NotImplementedError
NotImplementedError

meenchen, 2023-02-21:
Hi @729557989,

We did not have the full opset to support bull_bp setting since the required memory would not fit in the SRAM.

729557989, 2023-02-21:
Hi @meenchen 

Thanks for the reply! I have also tried sparse 49kbs ir (the minimal ran version) but this error still comes up.

lyc18, 2023-11-19:
> > Hi  你好@meenchen
> > Thanks for the reply! I have also tried sparse 49kbs ir (the minimal ram version) but this error still comes up.谢谢回复！我也尝试过稀疏 49kbs ir（最小 ram 版本），但仍然出现此错误。
> 
> Thanks for letting us know the issues. We will take a look into this.感谢您让我们了解问题。我们将对此进行研究。

Hi, is there a solution to this problem? I'm experiencing the same problem.
I followed the README steps in tiny-training/compilation, but the problem also occurs when I try to convert the generated model in json format to a C++ version via examples/tiny_training.py.