mit-han-lab/tinyengine #53: STM32CubeIDE version

byuljwa, 2023-02-14:
Hello, I tried the vww example with the stm32f746-disco board. However, as written, it works well in version 1.5.0, but I hope there is a place where I can find the reason and solution for not running in 1.11.0.

meenchen, 2023-02-14:
Hi @byuljwa,

We found that somehow the later version does not work with our camera driver, so that's the reason we use 1.5.0 in the tutorial. But everything else should work well in other versions.

byuljwa, 2023-02-17:
Thank you. camera driver mean ov2640.h? Which part do I need to edit?

meenchen, 2023-02-21:
> Thank you. camera driver mean ov2640.h? Which part do I need to edit?

We did not identify the root cause, so it is not clear which part needs to be edited.

byuljwa, 2023-02-24:
Thanks, I'll research more about this

closed 2023-04-18 by meenchen