mit-han-lab/tinyengine #57: No implemantation of convolve_1x1_s8_oddch_fpreq()

leaf82318, 2023-02-16:
Hello team.

I found that there is no kernel of convolve_1x1_s8_oddch_fpreq(),

Would you mind upload this kernel of tiny engine?

leaf82318, 2023-02-20:
I was trying to deploy a customized model by myself. And when run the code generation .py file ,it generated the code with the kernel of convolve_1x1_s8_oddch_fpreq(), but i cannot find the function withith the tinyengine. So the kernel of convolve_1x1_s8_oddch_fpreq() is ready?

729557989, 2023-02-21:
Hi @leaf82318 ,

Were you able to run the code generation.py file with your own customized model for on-device training?

If yes, did the 3 files you used to compile your customized model came from:

1. _graph.json_ file came under the _.model/testproj_ folder after you ran the _compilation/ir2json.py_ from MIT's **tiny-training** repo?
2. _params.pkl_ file came under the _.model/testproj_ folder after you ran the _compilation/ir2json.py_ from MIT's **tiny-training** repo?
3. _scale.json_ file came under the _ir_zoos/proxyless_quantize_ folder after you ran the _compilation/mcu_ir_gen.py_ from MIT's **tiny-training** repo?

Thanks!

leaf82318, 2023-02-21:
> Hi @leaf82318 ,
>
> Were you able to run the code generation.py file with your own customized model for on-device training?
>
> If yes, did the 3 files you used to compile your customized model came [...]

Thanks for your reply.

Recently, I just checked out the MIT's tinyengine repo, and run the vww.py with  my own customized model under the folder of examples. I have not try the MIT's tiny-training repo and know less about it. I'll try.

meenchen, 2023-02-21:
> I was trying to deploy a customized model by myself. And when run the code generation .py file ,it generated the code with the kernel of convolve_1x1_s8_oddch_fpreq(), but i cannot find the function withith the tinyengine. So the kernel of convolve_1x1_s8_oddch_fpreq() is ready?

Hi @leaf82318,

Thanks for reaching out. The `convolve_1x1_s8_oddch_fpreq` kernel is not implemented yet, but we will add the support soon.