mit-han-lab/tinyengine #65: Not implemented error got when trying to generate code for one of the already included networks

NepalLizarazu, 2023-03-10:
Hi, @meenchen, thanks for your great work. As I put in the title I got the following error (see image)

[image]

1. I started by following the tiny-training's repo compilation/README instructions, So: running the mcu_ir_gen.py and using the mcunet-5fps.pkl file.
2. Then I ran the ir2json.py file selecting the sparse_bp-49kb-1x3x128x128.ir just generated.
3. Then I took the scale.json from the ir_zoos corresponding folder (generated with mcu_gen_ir), the graph.json and the params.pkl from .model/testproj/ that were generated with ir2json, I put all three in the assets folder.

4. Finally I ran: python examples/tiny_training.py -f assets/sparse_bp-49kb-1x3x128x128-graph.json -D assets/sparse_bp-49kb-1x3x128x128-params.pkl -QAS assets/scale.json -m -g -d -FR
Making sure it was using the same 49kb sparse update scheme, and I got that Not implemented error, is there something that I might have missed during the process?

Pd. It seems to be originating from an "abs" operation that could not be handled, but being the provided examples in tiny-training makes me think I missed something at some point. Any thoughts on it?

meenchen, 2023-03-15:
Hi @NepalLizarazu, thanks for reporting this issue to us. We will take a look into it.

YanLi-LY, 2023-07-22:
Hi @meenchen @NepalLizarazu, have you solved this problem? I'm having the same error. If you've found a solution, could you share it with me? I'd really appreciate it.