mit-han-lab/tinyengine #67: About SE Block in tinyengine codes.

gunjupark, 2023-03-14:
Hi, I have some questions about tinyengine's codes.

While looking through tinyengine's codes, I found code in TfliteConvertor.py that handles SE Block. 
The code includes a comment that reads as follows:
`#         -> MEAN -> MEAN -> PWCONV -> PWCONV -> | ADD -> MUL ->     |
#  DWCONV                                        |            -> MUL |
#`

here are my questions:
1. Is it correct to assume that SE Block stands for Squeeze & Excitation module?

2. If so, does "ADD -> MUL -> MUL" refer to the h-swish activation function that replaces sigmoid?

Thank you

meenchen, 2023-03-15:
Hi @gunjupark,

> 1. Is it correct that SE Block means Squeeze & Excitation module?

Yes, `MEAN -> MEAN` is the pooling layer and `PWCONV -> PWCONV` is the two fc layers. 

> 2. If so, does "ADD -> MUL -> MUL" refer to the h-swish activation function that replaces sigmoid?

Yes, `ADD -> MUL -> MUL` is the activation function.

You can refer to this model: https://github.com/mit-han-lab/tinyengine/blob/main/assets/mcunetv2_large.tflite to check the architecture.

gunjupark, 2023-03-16:
I really appreciate you responding quickly to my question :)

I had missed that tflite model was there. 
Thanks you.

closed 2023-03-16 by gunjupark