mit-han-lab/tinyengine #68: Recipe for target 'firmware' failed

senceryucel, 2023-03-25:
Hello, 

Thanks for your job, everything is amazing. Following is what I am trying to face:

I have done everything correctly and it worked fine on my OpenMV H7 Board. However, when I try to run it on my OpenMV H7 Plus Board, I got an error at the last step of the building the firmware.

I changed TARGET=OPENMV4 to TARGET=OPENMV4P for both while building the source and while recompiling it. In building source part, everything works correctly. But I got the error below at the last step, while recompiling it:

```
 make[1]: Leaving directory '/home/senceryucel/Desktop/tinyengine/examples/openmv_person_detection/openmv/src/omv'
/usr/local/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: /home/senceryucel/Desktop/tinyengine/examples/openmv_person_detection/openmv/src/build/bin/firmware.elf section `.text' will not fit in region `FLASH_TEXT'
/usr/local/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: /home/senceryucel/Desktop/tinyengine/examples/openmv_person_detection/openmv/src/build/bin/firmware.elf section `.bss' will not fit in region `SRAM1'
/usr/local/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: section .dma_memory VMA [0000000030040000,0000000030043bff] overlaps section .bss VMA [0000000030000adc,0000000030042c8b]
/usr/local/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: section ._heap VMA [0000000030042c8c,000000003007ec8b] overlaps section .dma_memory VMA [0000000030040000,0000000030043bff]
/usr/local/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: section .d2_dma_memory VMA [0000000030043c00,0000000030047bff] overlaps section ._heap VMA [0000000030042c8c,000000003007ec8b]
/usr/local/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: region `SRAM1' overflowed by 257164 bytes
/usr/local/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: region `FLASH_TEXT' overflowed by 168120 bytes
collect2: error: ld returned 1 exit status
omv/ports/stm32/omv_portconfig.mk:649: recipe for target 'firmware' failed
make: *** [firmware] Error 1
make: Leaving directory '/home/senceryucel/Desktop/tinyengine/examples/openmv_person_detection/openmv/src'

```
What might be the cause for this? Thanks a lot in advance.

meenchen, 2023-03-27:
Hi @senceryucel,

This is because the compile fails to allocate the memory buffer required by TinyEgnine. You may need to disable some features that you don't need to for OPENMVP. For example, our patch disable some features to free the memory space: https://github.com/mit-han-lab/tinyengine/blob/b116d7dd57adb206193afa74118e1c6ac3e86237/examples/openmv_person_detection/openmv_person_detection.patch#L184
Also, we will need to reduce the size of framebuffer since we don't the high resolution in the demo:
https://github.com/mit-han-lab/tinyengine/blob/b116d7dd57adb206193afa74118e1c6ac3e86237/examples/openmv_person_detection/openmv_person_detection.patch#L351

Our code only takes care the configuration for OPENMV4, but you can apply similar changes to OPENMV4P as well. That should fix the problem.

senceryucel, 2023-04-03:
Thank you for your reply. I followed your steps, but now, I get "Error 2". I believe something else causes this error.

```
make[1]: Leaving directory '/home/senceryucel/Desktop/tinyengine/examples/openmv_person_detection/openmv/src/drivers/winc1500'
omv/ports/stm32/omv_portconfig.mk:593: recipe for target 'FIRMWARE_OBJS' failed
make: *** [FIRMWARE_OBJS] Error 2
make: Leaving directory '/home/senceryucel/Desktop/tinyengine/examples/openmv_person_detection/openmv/src'

```

How can I solve this? Thanks in advance.

closed 2023-04-14 by senceryucel