mit-han-lab/tinyengine #72: Recipe for target 'FIRMWARE_OBJS' failed - Error 2

senceryucel, 2023-04-12:
Hello, @meenchen 

I get "Error 2" while I am trying to build my OpenMV H7 Plus for your person detection. I could not solve this issue.

```
make[1]: Leaving directory '/home/senceryucel/Desktop/tinyengine/examples/openmv_person_detection/openmv/src/drivers/winc1500'
omv/ports/stm32/omv_portconfig.mk:593: recipe for target 'FIRMWARE_OBJS' failed
make: *** [FIRMWARE_OBJS] Error 2
make: Leaving directory '/home/senceryucel/Desktop/tinyengine/examples/openmv_person_detection/openmv/src'

```

How can I handle this? Thanks in advance.

meenchen, 2023-04-13:
Hi @senceryucel,

Can you provide more logs? or maybe the patch from git diff to reproduce this error? Hard to tell what's wrong based on the message.

senceryucel, 2023-04-14:
Hi @meenchen 

I believe the error 2 message caused by the git apply patch. Now, I am getting an error while trying to git apply. Here is what I get:

```
~/Desktop/tinyengine/examples/openmv_person_detection/openmv$ git apply ../openmv_person_detection.patch

../openmv_person_detection.patch:534: trailing whitespace.
CFLAGS += -std=gnu99 -Wall -Warray-bounds -mthumb -nostartfiles -fdata-sections -ffunction-sections 
../openmv_person_detection.patch:569: trailing whitespace.
	)	
error: patch failed: src/omv/boards/OPENMV4P/imlib_config.h:110
error: src/omv/boards/OPENMV4P/imlib_config.h: patch does not apply
error: patch failed: src/omv/boards/OPENMV4P/omv_boardconfig.h:150
error: src/omv/boards/OPENMV4P/omv_boardconfig.h: patch does not apply

```

I have tried it with ```git apply --reject --whitespace=fix``` options. When I do like that, I rejects some hunks and generates output as something like:
```
~/Desktop/tinyengine/examples/openmv_person_detection/openmv$ git apply --reject --whitespace=fix ../openmv_person_detection.patch
../openmv_person_detection.patch:26: trailing whitespace.
 
../openmv_person_detection.patch:534: trailing whitespace.
CFLAGS += -std=gnu99 -Wall -Warray-bounds -mthumb -nostartfiles -fdata-sections -ffunction-sections 
../openmv_person_detection.patch:569: trailing whitespace.
	)	
Checking patch src/hal/cmsis/include/arm_nnsupportfunctions.h...
Checking patch src/omv/Makefile...
Checking patch src/omv/boards/OPENMV4P/imlib_config.h...
error: while searching for:
// #define IMLIB_ENABLE_HIGH_RES_APRILTAGS

// Enable find_datamatrices() (26 KB)
#define IMLIB_ENABLE_DATAMATRICES

// Enable find_barcodes() (42 KB)
#define IMLIB_ENABLE_BARCODES

// Enable CMSIS NN
// #if !defined(CUBEAI)

error: patch failed: src/omv/boards/OPENMV4P/imlib_config.h:110
Checking patch src/omv/boards/OPENMV4P/omv_boardconfig.h...
error: while searching for:
// The maximum available fb_alloc memory = FB_ALLOC_SIZE + FB_SIZE - (w*h*bpp).
#define OMV_FFS_MEMORY          DTCM        // Flash filesystem cache memory
#define OMV_MAIN_MEMORY         SRAM1       // data, bss and heap memory
#define OMV_STACK_MEMORY        ITCM        // stack memory
#define OMV_DMA_MEMORY          SRAM2       // DMA buffers memory.
#define OMV_FB_MEMORY           AXI_SRAM    // Framebuffer, fb_alloc
#define OMV_JPEG_MEMORY         SRAM3       // JPEG buffer memory.
#define OMV_VOSPI_MEMORY        SRAM4       // VoSPI buffer memory.

#define OMV_FB_SIZE             (400K)      // FB memory: header + VGA/GS image
#define OMV_FB_ALLOC_SIZE       (100K)      // minimum fb alloc size
#define OMV_STACK_SIZE          (64K)
#define OMV_HEAP_SIZE           (236K)

#define OMV_LINE_BUF_SIZE       (3 * 1024)  // Image line buffer round(640 * 2BPP * 2 buffers).
#define OMV_MSC_BUF_SIZE        (2K)        // USB MSC bot data

error: patch failed: src/omv/boards/OPENMV4P/omv_boardconfig.h:150
error: while searching for:
#define OMV_DTCM_LENGTH         128K
#define OMV_ITCM_ORIGIN         0x00000000
#define OMV_ITCM_LENGTH         64K
#define OMV_SRAM1_ORIGIN        0x30000000
#define OMV_SRAM1_LENGTH        248K
#define OMV_SRAM2_ORIGIN        0x3003E000  // 8KB of SRAM1
#define OMV_SRAM2_LENGTH        8K
#define OMV_SRAM3_ORIGIN        0x30040000
#define OMV_SRAM3_LENGTH        32K
#define OMV_SRAM4_ORIGIN        0x38000000
#define OMV_SRAM4_LENGTH        64K
#define OMV_AXI_SRAM_ORIGIN     0x24000000
#define OMV_AXI_SRAM_LENGTH     512K

// Domain 1 DMA buffers region.
#define OMV_DMA_MEMORY_D1       AXI_SRAM
#define OMV_DMA_MEMORY_D1_SIZE  (8*1024) // Reserved memory for DMA buffers
#define OMV_DMA_REGION_D1_BASE  (OMV_AXI_SRAM_ORIGIN+(500*1024))
#define OMV_DMA_REGION_D1_SIZE  MPU_REGION_SIZE_8KB

// Domain 2 DMA buffers region.

error: patch failed: src/omv/boards/OPENMV4P/omv_boardconfig.h:175
Checking patch src/omv/modules/examplemodule.c...
Checking patch src/omv/ports/stm32/omv_portconfig.mk...
Applied patch src/hal/cmsis/include/arm_nnsupportfunctions.h cleanly.
Applied patch src/omv/Makefile cleanly.
Applying patch src/omv/boards/OPENMV4P/imlib_config.h with 1 reject...
Hunk #1 applied cleanly.
Rejected hunk #2.
Hunk #3 applied cleanly.
Applying patch src/omv/boards/OPENMV4P/omv_boardconfig.h with 2 rejects...
Rejected hunk #1.
Rejected hunk #2.
Applied patch src/omv/modules/examplemodule.c cleanly.
Applied patch src/omv/ports/stm32/omv_portconfig.mk cleanly.
```

I have done the corresponding changes for OPENMV4P instead of OPENMV4 in the file "openmv_person_detection.patch"
The changes I have done are:

```
diff --git a/src/omv/boards/OPENMV4P/imlib_config.h b/src/omv/boards/OPENMV4P/imlib_config.h
index fd395d87..262d527f 100644
--- a/src/omv/boards/OPENMV4P/imlib_config.h
+++ b/src/omv/boards/OPENMV4P/imlib_config.h
```

and

```
diff --git a/src/omv/boards/OPENMV4P/omv_boardconfig.h b/src/omv/boards/OPENMV4P/omv_boardconfig.h
index 412de472..f7da2c03 100644
--- a/src/omv/boards/OPENMV4P/omv_boardconfig.h
+++ b/src/omv/boards/OPENMV4P/omv_boardconfig.h
```

I have tried so hard and feeling like I am too close to the end for building it for H7 Plus. Your help is crucial for me, and I appreciate it in advance.

ps: There might be some problems caused by SRAM1, which I was not able to solve.

senceryucel, 2023-04-14:
Hi, @meenchen 

I found the cause behind the error. It is about the patch, like you have said.

I am opening a new issue about it since this issue is not an issue by itself.

closed 2023-04-14 by senceryucel