mit-han-lab/tinyengine #75: Is it possible to use a better resolution than QQVGA?

senceryucel, 2023-04-17:
Hey, @meenchen 

Like you have written in the person_detection_demo script,

```
sensor.set_framesize(sensor.LCD)  # Set frame size to QVGA 160x128
```

we made our inference on QQVGA (160x128):

```
sensor.set_framesize(sensor.QQVGA)
```

 (since we use the frame buffer instead of an LCD). It worked very nice. However, I tried to test it with higher resolutions but there did not occur any detection. I was wondering if it is possible to have a higher resolution with using this engine. Thanks a lot in advance.

meenchen, 2023-04-18:
Hi @senceryucel,

Glad to hear that works out. It is possible to use a higher resolution, but it may require some change in the memory allocation of openmv (i.e., the new patch https://github.com/mit-han-lab/tinyengine/pull/74/files) for the framebuffer and increases the memory footprint of TinyEngine. 

I would recommend changing the following resolution configurations, regenerating the code using `gen_code.sh`, and recompiling openmv, to see if your setting works out:
https://github.com/mit-han-lab/tinyengine/blob/d412e116593a2b26c8504ccfaadd727a770a4219/examples/detection_fpn.py#L28
https://github.com/mit-han-lab/tinyengine/blob/d412e116593a2b26c8504ccfaadd727a770a4219/examples/openmv_person_detection/openmv_person_detection.patch#L423

If you encounter some insufficient memory issue during compiling openmv, reducing the framebuffer in `omv_boardconfig .h` and try again.

meenchen, 2023-04-25:
Hi @senceryucel,

Look like the memory overflowed by a lot (~600KB). I am not sure we can fit it into the board (only 1MB SRAM in total). Can you share the resolution you use?

senceryucel, 2023-04-25:
Hi @meenchen 

I have tried it for 2 different resolutions. First, it was VGA (640x480). It threw the overflow with more than 600kBs, then I changed the resolution to QVGA (480x320). The outputs belong to the QVGA configs.

meenchen, 2023-04-26:
Hi @senceryucel,

I don't think the board is able to fit these 2 resolutions, at least not with the current memory scheduler of TinyEngine. Since our memory scheduler is completely decoupled with the framebuffer of the openmv firmware, it means we need to maintain two input images at the same time. This can easily cause memory overflow for such high resolution.