mit-han-lab/tinyengine #78: person_detection and face_mask_detection

Super-Karl, 2023-04-28:
Hello, may I ask what are the hardware devices and models of the demo OpenMV Cam H7? 
I want to repeat that. 
Thank you very much!!

meenchen, 2023-04-28:
Hi @Super-Karl,

Feel free to try our example demos on OpenMV Cam H7:
https://github.com/mit-han-lab/tinyengine/tree/main/examples/openmv_person_detection
https://github.com/mit-han-lab/tinyengine/tree/main/examples/openmv_vww

You can find the models here with the tflite format: https://github.com/mit-han-lab/tinyengine/tree/main/assets

Super-Karl, 2023-04-28:
Hi @meenchen ,
Thank you for your reply, I want to ask about the specific model of the development board and the model of the display screen that is required to implement the above demo.
I'm going to buy these hardware resources now, and I see that I may still need lithium batteries?

meenchen, 2023-04-28:
Sure, please check out the following links:

https://openmv.io/products/openmv-cam-h7