mit-han-lab/tinyengine #81: tutorial for inference did not work

lange-bleu, 2023-05-09:
I followed the Detailed Instruction,it went wrong at "python examples/vww.py"

[image]

meenchen, 2023-05-15:
Hi @lange-bleu,

I cannot reproduce this error on my side. Could you share more about your development environment?

lange-bleu, 2023-05-16:
hi @meenchen 
the figure below shows my env
I just run the Setup for Users in page https://github.com/mit-han-lab/tinyengine/tree/main
$conda create -n tinyengine python=3.6
$pip install -r requirements.txt
then I follow the Detailed Instruction in page https://github.com/mit-han-lab/tinyengine/tree/main/tutorial/inference

[image]