mit-han-lab/tinyengine #100: No module named 'code_generator'

tracyyy19, 2024-03-26:
So, I tried following the instructions of this tutorial, https://github.com/mit-han-lab/tinyengine/tree/main/tutorial/inference but I still get the same error

(tinyengine) C:\Users\USER\Downloads\tinyengine-main\tinyengine-main\examples>python vww.py
Traceback (most recent call last):
  File "vww.py", line 19, in <module>
    from code_generator.CodegenUtilTFlite import GenerateSourceFilesFromTFlite
ModuleNotFoundError: No module named 'code_generator'

jin3qin, 2024-05-16:
Before running the Python script, you need to ensure that it runs in a conda environment with the configuration name **tinyengine**. If the "code_generator" module is missing, you need to run` export PYTHONPATH=${PYTHONPATH}: $(tinyengine's path)` first.

jin3qin, 2024-05-16:
> So, I tried following the instructions of this tutorial, https://github.com/mit-han-lab/tinyengine/tree/main/tutorial/inference but I still get the same error
> 
> (tinyengine) C:\Users\USER\Downloads\tinyengine-main\tinyengine-main\examples>python vww.py Traceback (most recent call last): File "vww.py", line 19, in from code_generator.CodegenUtilTFlite import GenerateSourceFilesFromTFlite ModuleNotFoundError: No module named 'code_generator'

Before running the Python script, you need to ensure that it runs in a conda environment with the configuration name **tinyengine**. If the "code_generator" module is missing, you need to run `export PYTHONPATH=${PYTHONPATH}: $(tinyengine's path)` first.